rakeshpai/pi-gpio #32: opening for output may be causing a change in GPIO state (to 0) occasionly

markv1, 2015-01-09:
when running the following code repeatedly (setting the GPIO pin to 1 -relay off multiple times), occationally i hear the relay click and get a momentary contact on.  I also see a GPIO-Admin: could not flush data to /sys/class/gpio/export: device or resource busy.   I am using a RPb+ and sainsmart 8ch relay board.   The momentary contact on when the relay is already off is the most problematic.

 gpio.open(pin,'output', function(err) {
         gpio.write(pin,highLow,function() {
                gpio.close(pin);      
      });
 });

lucasmoglia, 2015-02-10:
Hi! Did you find any solution to this problem? Because i'm having the same trouble. I'm using a setTimeout function but the pin gets up just for a second or less.

skoold2003, 2015-06-13:
Are you referring to the fact that the pin value is set to 0? I added a line to the end of the set direction method as shown here:

fs.writeFile(sysFsPath + "/gpio" + pinMapping[pinNumber] + "/value", 1, (callback || noop));

This sets the default export value to 1. This should be added to the library IMO but I'm not sure how to request that.