rakeshpai/pi-gpio #38: Is gpis-admin still needed?

aramperez, 2015-05-15:
I've just installed 2015-05-05-raspbian-wheezy and I'm getting the following error (even though the example still runs correctly):
"Error when trying to open pin xx
/bin/sh: 1: gpio-admin: not found"

I have not installed gpio-admin.

Thanks,
Aram

omkarkhair, 2015-07-08:
I get the same error when I try to run my app at boot from `/etc/init.d`.

aramperez, 2015-07-08:
It appears that no one is monitoring this list :-(

rakeshpai, 2015-07-08:
I don't have a Pi handy to test, so I can't confirm. Recent pull requests
have changed things around a bit. I think that gpio-admin is still needed,
though.

omkarkhair, 2015-07-08:
My issue was resolved after changing `gpioAdmin = "gpio-admin"` to `gpioAdmin = "/usr/local/bin/gpio-admin"`. Alias apparently don't work in init.d scripts.

tjanson, 2015-07-08:
You could also try [my branch](https://github.com/rakeshpai/pi-gpio/tree/gpioutil-fs-hybrid) (Pull Request: #27, original issue #25), which replaces quick2wire’s outdated `gpio-admin` with Wiring Pi’s `gpio` utility.

However, I haven’t used this since last fall, so it’s possible that something broke in the mean time. Feel free to give it a try, though, and submit feedback (separate issue please). I *will* get back to this eventually, I just don’t know when …

aramperez, 2015-07-09:
@tjanson, is there a way to "assume" the pins have already been exported and modes set? I see there is an optional 'exportMode' argument that can be passed but I don't want to be passing 'off' everytime I call an API. I've written a utility (that uses wiring-pi) that reads a JSON file to configure and export the GPIO pins, including setting the pull-up/down resistors and whether the pin is active low/high.

tjanson, 2015-07-09:
@aramperez Other than that parameter, no. You could adapt the code or wrap the call so that 'off' is the default. Perhaps a `setDefaultExportMode` function or a config file would be nice. Feel free to add whatever suits you.

aramperez, 2015-07-09:
@tjanson I'll let you know what I end up using, thanks.

closed 2015-08-26 by tjanson