rakeshpai/pi-gpio #41: failed to change group ownership of /sys/devices/virtual/gpio/gpio23/direction: No such file or director

born2net, 2015-08-26:
Hello,
I installed pi-gpio on latest Raspian:

root@raspberrypi:~# uname -a
Linux raspberrypi 4.1.6-v7+ #810 SMP PREEMPT Tue Aug 18 15:32:12 BST 2015 armv7l GNU/Linux

tried running the sample script both as root and also installed the admin tool and still I get the same error on the latest revision Raspberry pi 2

Error when trying to open pin 16
gpio-admin: failed to change group ownership of /sys/devices/virtual/gpio/gpio23/direction: No such file or director

any help is appreciated

regards

Sean

tjanson, 2015-08-26:
Did you follow the [installation instructions], in particular installing `quick2wire-gpio-admin`?

born2net, 2015-08-26:
yes I did follow and did add the quick2wire-gpio-admin
but still same error :(

tjanson, 2015-08-26:
See the issues I linked above, those people apparently encountered the same problem (and possibly solved it).

It’s definitely a `gpio-admin` problem, and that program has been abandoned for several years. That’s why I made some effort to switch to a different tool, see #27 – you can [give that branch a try](https://github.com/rakeshpai/pi-gpio/tree/gpioutil-fs-hybrid). (It worked fine when I wrote it, but I haven’t used it recently.)

closed 2015-08-26 by tjanson