rakeshpai/pi-gpio #45: correct test/pi-gpio.js

alex-dacosta, 2015-09-29:
you should also add:
var 
	sysFsPathOld = "/sys/devices/virtual/gpio", // pre 3.18.x kernel
	sysFsPathNew = "/sys/class/gpio", // post 3.18.x kernel
	sysFsPath;

// tests the device tree directory to determine the actual gpio path
if (fs.existsSync('/sys/devices/soc')) {
	sysFsPath = sysFsPathNew;
} else {
	sysFsPath = sysFsPathOld; // fallback for old kernels
}
and use `sysFsPath` in paths

tjanson, 2016-02-28:
You are correct, and that looks good. Feel free to submit a PR. Unfortunately this project is for the time being somewhat abandoned: Neither the owner nor I currently use it.

tjanson, 2016-02-28:
Wait, unless I’m missing something this has been fixed by https://github.com/rakeshpai/pi-gpio/commit/82ef7bddbb0af8edf03470023be71febeac6ed32 since March 2015 (in `master`, not the other branch).