rakeshpai/pi-gpio #47: gpio-admin: failed to change group ownership of /sys/devices/virtual/gpio/gpio10/direction: No such file or directory

Andrew5569, 2015-10-04:
Every time I try to write to an output I get the following error "gpio-admin: failed to change group ownership of /sys/devices/virtual/gpio/gpio10/direction: No such file or directory"

Code is as follows: 

var gpio = require('pi-gpio')

var led = 19

gpio.open(led, "output", function(err) {
  gpio.write(led, 1, function() {
      gpio.close(led)
  })
})

trojanspike, 2015-11-11:
@Andrew5569 
In file : /boot/config.txt
add : device_tree=

In file : /etc/udev/rules.d/91-gpio.rules
add : KERNEL=="gpio*", RUN="/bin/sh -c 'chgrp -R gpio /sys/%p /sys/class/gpio && chmod -R g+w /sys/%p /sys/devices/virtual/gpio'"

Note , unsure that your user is part of group "gpio"

tjanson, 2016-02-28:
May be due to #45. I’m closing this as stale but feel free to reopen.

closed 2016-02-28 by tjanson