rakeshpai/pi-gpio #53: About usage on raspberry 3B

attacking, 2016-04-06:
I tried my best to control my 3B's GPIO by pi-gpio, however it has never worked. （I have successfully controlled the GPIO by python， so the question is not the GPIO）
I want to know if the pi-gpio and  quick2wire-gpio-admin apply to controlling 3B.

rakeshpai, 2016-04-06:
Could you give some more details? What were you trying, and what error did
you see?

attacking, 2016-04-06:
1. I connect the raspberry 3B(with RASPBIAN JESSIE system) by SSH(PuTTY).
2. Then I install the node and npm successfully.
3. Since it suggests that we'd better to install "quick2wire-gpio-admin", I do.(After this process nodejs still warned that I had  no access to the gpio, finally fixed by adding chip models).
4. After these steps, my code as follows:

```javascript
var gpio = require('pi-gpio')
var errHandler = function (err){
  console.log(err);
}
gpio.open(40, "output", errHandler )
while (1){
  gpio.write(40, 0, errHandler)
  console.log("Pin-40 : 0")
  for(var i=0;i++;i<500000){}
  gpio.write(40, 1, errHandler)
  console.log("Pin-40 : 1")
  for(var j=0;j++;j<250000){}
}
```

Phenomenon: these console.log statements was executed, however voltage of pin-40 did not change and had nothing other.

tjanson, 2016-04-09:
The underlying problem here is that Quick2Wire is really quite abandoned, it hasn’t been updated in 3 (!) years. This makes pi-gpio fundamentally unstable as long as it’s still based on Q2W. See PR #27 for some more details – the bottomline is: to keep up with new versions of the Raspberry Pi (and Node, etc.), pi-gpio needs to switch to another utility (as I did with that PR), as well as general maintenance (no one is doing that, as far as I can tell – I’m certainly not).

attacking, 2016-04-09:
it's right. And I gave up at last. Thanks, tjanson.

closed 2016-04-09 by attacking