rakeshpai/pi-gpio #54: Not Working on Raspberry Pi 2 Model B

chelinho139, 2016-05-26:
Hello, I have followed the guide step by step.
But when I try to use this wrapper I get te following error message:

**Error when trying to open pin 16
gpio-admin: failed to change group ownership of /sys/devices/virtual/gpio/gpio23/direction: No such file or directory**

I cant open any pins...

I am using the latest version of  quick2wire.

The raspberry however, is working fine. I can read/write pins using the terminal and the command: "gpio read 16"

Any ideas?

gnut3ll4, 2016-06-30:
I have the same issue, I can manage to make quick2wire work on bash command line but this still doesn't work in node.js.

What I did is changing the gpio system path in quick2wire and make + install : 

`diff --git a/src/gpio-admin.c b/src/gpio-admin.c
index 38569bb..faea863 100644
--- a/src/gpio-admin.c
+++ b/src/gpio-admin.c
@@ -27,7 +27,7 @@ static void usage_error(char **argv) {
 
 static void allow_access_by_user(unsigned int pin, const char *filename) {
   char path[PATH_MAX];
-  int size = snprintf(path, PATH_MAX, "/sys/devices/virtual/gpio/gpio%u/%s", pin, filename);
+  int size = snprintf(path, PATH_MAX, "/sys/class/gpio/gpio%u/%s", pin, filename);
   
   if (size >= PATH_MAX) {
     error(7, 0, "path of GPIO pin is too long!");
`

DaniyalGeek, 2016-07-19:
it works for me must just say good job, it works! I am running a pi b +, with recent firmware etc, Did some manual changes to the file that was changed according to this commit, namely the file gpio-admin.c which can be found within the folder structure quick2wire-gpio-admin/src/gpio-admin.c and replaced the path as described in files changed (look at the tab files changed to get a visual). Did a "sudo make uninstall" from within the quick2wire-gpio-admin folder, then "sudo make install" and added the user "sudo adduser $USER gpio" (where $USER is your user..). Logout and login, and then voila it works! :) Just wanted to overexplain some stuff here incase somebody was struggling.